DeepSOIC/Lattice2 #2: Macro not recorded for Explode Compound

PatONeil, 2016-01-23:
When recording a macro and using Lattice2 Explode Compound, the operation is not recorded.   Is there any manual way to add this code to a macro?  If so please send to me at oneilpatr at gmail dot com

Thanks

DeepSOIC, 2016-01-23:
Hi!

There was some problem with coding it... I'll take another look.

A quick way to include that into a macro is

    Gui.Selection.clearSelection()
    Gui.Selection.addSelection(App.ActiveDocument.ActiveObject)
    Gui.runCommand("Lattice2_Explode")

PatONeil, 2016-01-23:
Thanks for the quick response.   I needed to modify the selection statement
but other than that, it works great.

 

Pat

PS:  This is a great plug-in.  I don't know how I would have created my part
without it.

 

  _____  

 

<https://github.com/DeepSOIC/Lattice2/issues/2#issuecomment-174169469>  it
on GitHub.
<https://github.com/notifications/beacon/AF-KkCe-Pa7f5bhZjOot3vCXPZwpSiW4ks5
pc08tgaJpZM4HK2hj.gif>

commit a750a8e7 (2016-01-24): CompoundFilter: Explode: fix #2 + group together the result of explosion
closed 2016-01-24 by DeepSOIC