DeepSOIC/Lattice2 #13: Error with FreeCAD 0.14

nvdl, 2016-10-11:
https://github.com/DeepSOIC/Lattice2/blob/master/lattice2Common.py#L54

I don't know if FreeCAD 0.14 is supported or not.

Gives an error with FreeCAD 0.14:

Error: 'module' object has no attribute 'FreeCADError'

FreeCAD:

OS: Debian GNU/Linux 8.6 (jessie)
Word size: 64-bit
Version: 0.14.3702 (Git)
Branch: releases/FreeCAD-0-14
Hash: b3368125c63289ec8ce9faec2b2ae4c78d436406
Python version: 2.7.8
Qt version: 4.8.6
Coin version: 4.0.0a
SoQt version: 1.6.0a
OCC version: 6.7.0

DeepSOIC, 2016-10-11:
FreeCAD v0.14 is not supported, sorry. 0.14 is ancient.

DeepSOIC, 2017-02-15:
There is such a mention already:
> Lattice2 WB requires FreeCAD no less than v0.16.5155.
In case you are wondering, that is when dropdown toolbar buttons became supported by FreeCAD.