DeepSOIC/Lattice2 #16: Does Lattice2 have anything to do with FC mantis feature 0001262

luzpaz, 2017-05-11:
[Link to 1262](https://freecadweb.org/tracker/view.php?id=1262)

DeepSOIC, 2017-05-11:
> Does Lattice2 have anything to do with FC mantis feature 0001262

Not much. I think the suggested "Lattice" feature is impractical, because it will generate a very heavy and complex b-rep, that will not help at all in the end because it will simply confuse slicers. 

I think such a feature can make sense only in tight integration of some slicer, in a form of creating a compsolid-like geometry.

luzpaz, 2017-05-13:
> Not much. I think the suggested "Lattice" feature is impractical, because it will generate a very heavy and complex b-rep, that will not help at all in the end because it will simply confuse slicers.

> I think such a feature can make sense only in tight integration of some slicer, in a form of creating a compsolid-like geometry.

@DeepSOIC thank you. Would you mind notating that ticket with you opinion? If and when you have a moment..? Thanks!

closed 2018-08-24 by DeepSOIC