DeepSOIC/Lattice2 #17: Lock and unlock icons greyed out, FreeCAD V0.16 build 6712

perkmeister, 2018-03-08:
I have just installed Lattice2 (thanks for this!) with FreeCAD V0.16 build 6712 but it appears that the lock and unlock recompute buttons are permanently greyed out. This has been confirmed by another user. Possible regression?

DeepSOIC, 2018-03-08:
You should make a recompute locker object (a button nearby). Then you should be able to use recompute lock/unlock buttons.
I can't create one, so it is a regression. I'm not sure if I had disabled it on purpose... let me check...

DeepSOIC, 2018-03-08:
Yep, I accidentally committed unfinished code in this commit:
3f568ed5700f5bcfe6347c1bc62d168fa9b6cdc0

perkmeister, 2018-03-08:
OK, thanks!

perkmeister, 2018-03-11:
Any idea when a version that has the recompute functionality will be committed? I only need those functions for now. Thanks!

DeepSOIC, 2018-03-12:
Hopefully, soon.
BTW, you can use FC v0.17, which has some of these functions built-in now.

perkmeister, 2018-03-12:
OK, I'm nearing the end of a big project so can't investigate v0.17 at the moment. I've got a hack which is to add a part and create a loop in the dependency graph to create a non-DAG, and manually remap one particular sketch to the part that creates a loop, or one that doesn't. This I think is a maybe a somewhat simplified version of what you do.

DeepSOIC, 2018-03-12:
Should be fixed now, please update your Lattice2.

closed 2018-03-12 by DeepSOIC

perkmeister, 2018-03-12:
Thanks for the quick response. Unfortunately this doesn't fix it on my system. The option to create a locker object is now not present in the Recompute menu (it was in the last commit but greyed out), and there is no button.

OS: Windows 7
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.16.6712 (Git)
Build type: Release
Branch: releases/FreeCAD-0-16
Hash: da2d364457257a7a8c6fb2137cea12c45becd71a
Python version: 2.7.8
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 6.8.0.oce-0.17

DeepSOIC, 2018-03-12:
4d4ac43, please try again!

perkmeister, 2018-03-13:
Great, that works! Thank you very much for sorting that out! 

Could you possibly clarify the operation of the two recompute buttons? One says "Recompute document: recomputes the document, ignoring that recomputes are locked", and the other is "Force recompute: recomputes all features in the document". 

Does the first one only start recomputing when it sees a touched component and then recomputes everything dependent on it, and force recompute recomputes everything? Or is this an object by object locking feature that's supposed to recompute everything except locked objects? How do you lock individual objects, or is this work in progress? Thanks!

DeepSOIC, 2018-03-13:
*recompute document*: just like regular FreeCAD's recompute button, recomputes only stuff that needs to be recomputed. Except, it bypasses the locking.

*force recompute*: recomputes just everything in your project, might take a while.

perkmeister, 2018-03-13:
Right, got it. Thanks again for the fix ;)

commit 02ac4b24 (2018-03-12): RecomputeLocker: fix #17 ; upgrade for 0.17

* Fixes #17 "lock and unlock icons greyed out"
* add support for FC0.17's built-in recompute contol.