DeepSOIC/Lattice2 #19: polar array - cannot change Num Elements

herrdeh, 2018-07-10:
I can select the number field - but the number of elements will not change, but always remain "5". - Arrows right hand side are grayed out.

OS: Ubuntu 18.04 LTS
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.14061 (Git)
Build type: None
Branch: master
Hash: c4fc02cbcfff975712e977dc08f859fba71ba0ad
Python version: 2.7.14
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.2.0
Locale: German/Germany (de_DE)

Great WB, bthw. Can really do awesome things!

Cheers,
Wolf

DeepSOIC, 2018-07-10:
Hi!
"Num Elements" property is read-only, and shows the actual number of placements in an array. Use Count property, in "Lattice Series Generator" section.

herrdeh, 2018-07-10:
True! - Thanks for the quick hint. 
A little confusing, as "Num Elements" comes first in the input list.

closed 2018-07-10 by herrdeh