DeepSOIC/Lattice2 #24: PolarArray Values Property losing delimiters?

dtcomp, 2018-09-02:
After setting Values Source to "Values Property", editing Values list,
then clicking "OK", the list is collapsed into one value.
Initially, the list appears good, only after editing is it corrupted.
Oddly, editing the corrupted list by separating the values with `<CR>`
has no effect.
AND, If corrupted list is edited adding only a comma (no `<CR>`) between values, list is stored properly again.

I'm using freecad-daily...

OS: Ubuntu 16.04.5 LTS
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.14498 (Git)
Build type: Release
Branch: master
Hash: e381e6b739a6b83b5f476546b64569cfa03a4e3a
Python version: 2.7.12
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)

DeepSOIC, 2018-09-02:
Hi!
Maybe a consequence of a recent FreeCAD fix?
https://forum.freecadweb.org/viewtopic.php?f=22&t=30462

Anyway, I'll check what's going on in a few days.

dtcomp, 2018-09-03:
Yes, that seems likely..
FWIW, I can reproduce on Windows as well:

OS: Windows 10
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.14495 (Git)
Build type: Release
Branch: master
Hash: c717d1121d888fe6cfbd8971b5f59311a34b37bc
Python version: 3.6.6
Qt version: 5.6.2
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)

DeepSOIC, 2018-09-03:
should be fixed by freecad commit https://github.com/FreeCAD/FreeCAD/commit/9e5b31fe2a49c57436fb47df7b56150db6362dcd
I didn't test.

closed 2019-08-27 by DeepSOIC