DeepSOIC/Lattice2 #25: No Lattice Update/Recompute?

dtcomp, 2018-09-02:
It seems sometimes my PolarArray gets "stuck" -
varying parameters, etc. result in no change to Lattice objects or display...
The way to effect a change is to delete my Lattice and re-create.

DeepSOIC, 2018-09-02:
No idea. Can you share a project (if the state survives through save-load, of course)?

DeepSOIC, 2018-09-02:
Also, do you get any messages in report view? Does it have a checkmark or exclamation mark olerlaid on the icon in tree view?

dtcomp, 2018-09-03:
[https://github.com/dtcomp/L2test](url)
It seems to survive on 3 platforms here, including windows...
Nothing shows in report view and no warning icons treeview.

dtcomp, 2018-09-03:
It could be that I just don't know how to use FreeCad, just starting!

dtcomp, 2018-09-03:
I do see a console message (starting from terminal) :
`Populate: Placment is not a placement or an array of placements. Results may be unexpected.`

And, i do notice that changes to "Placement:Attachment Offset" appear immediately.

DeepSOIC, 2018-09-03:
OK, I see that your project has some broken Lattice2 objects. They somehow lost their python proxies (a python-powered feature is a C++ object with a python object "proxy" bound to it. Proxy contains the actual Lattice code).

I don't know how did it happen. The only possible way I can think this might have happened is if you opened this file in a FreeCAD that has no Lattice2 installed (or the installation was broken), and saved it.