DeepSOIC/Lattice2 #28: Memory Leak

sliptonic, 2019-05-15:
I think Lattice2 has a memory leak.  I left a model open over night and my machine was almost unusable in morning.  I have 16GB and FreeCAD had almost all of it. 

I closed and reopened FreeCAD and the model and the memory started to creep up immediately.  After playing with it some, I've observed the following:

1) It doesn't start consuming memory just by loading the workbench
2) It's related to a placement object.  Once a placement is added, it starts to leak.
3) More placements == faster leak.
4) closing the document doesn't release the memory.  Have to close FreeCAD.  

I created a simple rectangular pad, added a placement to the corner and it leaks roughly 3MB/Minute

Here's my FreeCAD info.  This is basically the master branch as of yesterday with a some python only changes in Path. I can re-verify with a clean build of master if necessary. 

OS: Debian GNU/Linux buster/sid (i3/i3)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.16726 +1 (Git)
Build type: Unknown
Branch: (HEAD detached at mlampert/feature/4th-axis-setup)
Hash: 599ed42f7968cb90811e73e94c7c60ead7ee3484
Python version: 3.7.3rc1
Qt version: 5.11.3
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)

DeepSOIC, 2019-05-15:
I'll test.
Any other addons installed like Part-o-magic? That one could easily leak, as it continuously runs some code,  tracking for changes. In Lattice, the only thing that runs continuously are command's isActive methods, which FreeCAD calls periodically for everything. So if there is a memory leak, it's likely coming from isActive() code.

DeepSOIC, 2019-05-15:
So far, after 4 minutes of sitting there doing nothing, memory allocated by FC dropped from 201 MB to 200 MB. Like you said, I made a rectangular Pad, and attached a Placement to a corner with "Translate origin" mode.

DeepSOIC, 2019-05-15:
Well, I created 1000 placements by exploding a polar array. Now I can see it leaking in real time, about 0.5 MB/s

sliptonic, 2019-05-15:
No Part-o-Magic.  I've got A2plus installed but not using it in the model or even opening the workbench. Can test without it but it sounds like you're on it already.

DeepSOIC, 2019-05-15:
I now tried Part Explode of a Draft Array (1000 cubes). Same memory creep. Didn't even open Lattice2. However, Part Explode is actually using Lattice2 (Part CompoundFilter is Lattice CompoundFilter with slight modifications). Investigation continues...

DeepSOIC, 2019-05-15:
Made 1000 pure Part Cubes.

    for i in range(1000):
      App.ActiveDocument.addObject('Part::Box')

still creeps up =(
```

DeepSOIC, 2019-05-15:
Disabled all addons. No creep...

DeepSOIC, 2019-05-15:
Okay, the creep I am observing clearly comes from Part-o-magic. I pause its observer, and the creep stops. Now, re-enabling other add-ons...

DeepSOIC, 2019-05-15:
With PoM observer paused, I'm not seeing any leak with my Lattice2 explode-array-of-1000 test.

sliptonic, 2019-05-15:
Some of the leaking I'm seeing may be Path related.  I did the 1000 cube test above and saw no leak. Switching to Path wb and the leak started.  Will investigate.

luzpaz, 2022-10-15:
@sliptonic did anything ever come of this ?