DeepSOIC/Lattice2 #35: FC 0.19 qRegisterResourceData signature mismatch?

dtcomp, 2019-09-22:
I'm trying out 0.19 and got some errors loading Lattice2.

OS: Ubuntu 18.04.3 LTS (ubuntu:GNOME/ubuntu-wayland)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.
Build type: Release
Python version: 3.6.8
Qt version: 5.9.5
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)

Loading Part module... done
'PySide2.QtCore.qRegisterResourceData' called with wrong argument types:
  PySide2.QtCore.qRegisterResourceData(int, str, str, str)
Supported signatures:
  PySide2.QtCore.qRegisterResourceData(int, bytes, bytes, bytes)
Traceback (most recent call last):
  File "<string>", line 49, in Initialize
  File "/home/mike/.FreeCAD/Mod/lattice2/Lattice2.py", line 28, in <module>
    import Lattice2CodeModules as Code
  File "/home/mike/.FreeCAD/Mod/lattice2/Lattice2CodeModules.py", line 1, in <module>
    import lattice2Common as _common_code
  File "/home/mike/.FreeCAD/Mod/lattice2/lattice2Common.py", line 88, in <module>
    import lattice2_rc
  File "/home/mike/.FreeCAD/Mod/lattice2/lattice2_rc.py", line 21, in <module>
    qInitResources()
  File "/home/mike/.FreeCAD/Mod/lattice2/lattice2_rc.py", line 16, in qInitResources
    QtCore.qRegisterResourceData(0x01, qt_resource_struct, qt_resource_name, qt_resource_data)

dtcomp, 2019-09-22:
After manually install latest master,
a different error appears, and the previous is gone...

Now:

Loading Part module... done
No module named 'replaceobj'
Traceback (most recent call last):
  File "<string>", line 55, in Initialize
  File "/home/mike/.FreeCAD/Mod/Lattice2/Lattice2.py", line 36, in <module>
    import Lattice2GuiTools as GuiTools
  File "/home/mike/.FreeCAD/Mod/Lattice2/Lattice2GuiTools.py", line 3, in <module>
    import lattice2SubstituteObject  as SubstituteObject
  File "/home/mike/.FreeCAD/Mod/Lattice2/lattice2SubstituteObject.py", line 29, in <module>
    from replaceobj import replaceobj #from OpenSCAD wb, the code that drives replaceChild

dtcomp, 2019-09-23:
This could be a misconfiguration issue on my system...

DeepSOIC, 2019-09-23:
I have recently messed with the resource file, it may have been the cause. But I am not having the problem.

Do you have OpenSCAD workbench on your FreeCAD?

DeepSOIC, 2019-09-23:
Whatever the cause of the problem is, I should make Lattice2 more tolerant to these errors. While they are serious, they shouldn't stop the whole workbench from working.

dtcomp, 2019-09-23:
OpenSCAD and most everything else was not loading,
Now, after playing with py2 and py3 updates and such, Almost everything
is loading and Lattice2 is working!
Probably a non-issue so I will close..

closed 2019-09-23 by dtcomp