DeepSOIC/Lattice2 #40: Attachment using objects name instead of label

leoheck, 2020-04-19:
Hi, I am experimenting with this lattice2 plugin. Thanks, by the way, it is awesome.

I am using the polar array, and in the Attachment panel I am seeing this `Pad004` but I don't have any Pad004, this name is hidden from the user. What I have is `wall_ribs` instead. This is the name that the plugin should use. What happens it most of the time the user does not rename features so de label has the same name as the object. 

Ah, and this should be used everywhere if you agree that this is an important change.

DeepSOIC, 2020-04-19:
well, attachment panel displays internal name for a reason: it is unique.

leoheck, 2020-04-19:
Yes, I see. It should be using it internally, but it can display a label for the user. Or, even better, a `label (name)` pair.

leoheck, 2020-04-20:
When I clicked in the Body, I saw this. Maybe following the same pattern (seen in 0.19) may be even better.

[image]

DeepSOIC, 2020-04-20:
> It should be using it internally, but it can display a label for the user. 

The fact that the field is editable gets in the way (i.e. you can copy-paste these refs from one field to another, or modify index). It is of course doable (e.g. the text is modified as soon as you put a cursor in the field), but it's complicated.

leoheck, 2020-04-20:
I see you are right, you have a good point.
My last attempt to help users, maybe splitting this field, with the name being able to edit and the label being displayed on its side?