DeepSOIC/Lattice2 #41: Z-Placement in PolarArray

leoheck, 2020-04-19:
I am changing the Z placement of a PolarArray feature but it keeps resetting it to 0. Is it right?

The features are not being generated too (maybe because of the wrong z-placement). So, I am adding z=18, for instance, but after the refresh, it will be moved to z=0 again.

Am I doing something wrong or this is a bug?

The base feature is the small whitish trapezoidal form (top-left)
[image]

Current settings are these.
[image]

DeepSOIC, 2020-04-19:
> I am changing the Z placement of a PolarArray feature but it keeps resetting it to 0. Is it right?

If the array is attached, and you are changing Placement's Z (not AttachmentOffset Z) - yes. But the fact that it's not grayed out is a bug.

DeepSOIC, 2020-04-19:
> The features are not being generated too

can you share a file, please? I can't guess what's wrong here

leoheck, 2020-04-20:
Sure, here it is (but I think you already saw it in Freecad Forum)
[lattice_polar_array.zip](https://github.com/DeepSOIC/Lattice2/files/4500589/lattice_polar_array.zip)

I was not using the latest version of Lattice2, now I am using, and I am seeing this
```
Traceback (most recent call last):
  File "/home/lheck/.FreeCAD/Mod/lattice2/lattice2BaseFeature.py", line 148, in execute
    plms = self.derivedExecute(obj)
  File "/home/lheck/.FreeCAD/Mod/lattice2/lattice2PolarArray2.py", line 119, in derivedExecute
    self.updateReadonlyness(selfobj)
  File "/home/lheck/.FreeCAD/Mod/lattice2/lattice2PolarArray2.py", line 103, in updateReadonlyness
    arc = self.fetchArc(selfobj) 
  File "/home/lheck/.FreeCAD/Mod/lattice2/lattice2PolarArray2.py", line 115, in fetchArc
    return fetchArc(lnkobj, sub)
  File "/home/lheck/.FreeCAD/Mod/lattice2/lattice2PolarArray2.py", line 56, in fetchArc
    linkedShape = obj.Shape
<class 'AttributeError'>: 'App.GeoFeature' object has no attribute 'Shape'
448.595 <App> Document.cpp(3715): Failed to recompute lattice_polar_array#PolarArray: 'App.GeoFeature' object has no attribute 'Shape'
Recompute failed! Please check report view.
```

DeepSOIC, 2020-04-20:
> now I am using, and I am seeing this..

well, that's because the array is attached to XY plane. The array is trying to read out arc range from the shape it is attached to, and that causes the error (the plane having no shape, particularly).

leoheck, 2020-04-20:
Like this?

[image]

Nothing has changed.

DeepSOIC, 2020-04-20:
Well, the error is gone at least.

DeepSOIC, 2020-04-20:
I think the problem with your model may be a tangent intersection in a Boolean operation. Try changing the rib sketch a bit and see if the situation is resolved. Like so, for example:
[image]

leoheck, 2020-04-20:
Nothing so far. The shell is a BSPline, does it can be related?

What should I put in Radius? The tip of the arrows should be inside the feature? Or the base of the feature? Or something else?

leoheck, 2020-04-20:
Issue solved, It was missing the LatticePattern feature.

closed 2020-04-20 by leoheck