DeepSOIC/Lattice2 #45: Type error with spredsheet data in FreeCAD 0.19

shyblower, 2020-04-26:
When I try to use a value from a FC spreadsheet for the "Count" property of e.g. the PolarArray then the following exception is thrown:
`<Exception> Invalid type 'l'
in property binding 'Count'
11942.7 <App> Document.cpp(3693): Failed to recompute ArrayTest#PolarArray: Invalid type 'l'
in property binding 'Count'`

it does work in FC 0.18, though.

I also figured out a hack to work around this issue in FC 0.19. When I add a unit symbol like "mm" at the end of the number in the spreadsheet, then I can use it for the "Count" property. It will only issue a warning that the unit symbol will be stripped.

DeepSOIC, 2020-04-26:
I can reproduce. Seems to be a freecad bug. I get this error:

`<Exception> Invalid type 'long'
in property binding 'Count'`

if I set a fractional value like 4.000000001, the error disappears.

shyblower, 2020-04-26:
Yes, fractional values also work for me, just tried :)

DeepSOIC, 2020-04-26:
https://forum.freecadweb.org/viewtopic.php?f=3&t=45720

DeepSOIC, 2020-04-27:
Should be fixed since FC 0.19.20814

closed 2020-04-27 by DeepSOIC