DeepSOIC/Lattice2 #47: Multiple colors/appearances

yankee14, 2020-05-02:
Thank you for this extremely useful tool.

I am working on making a 3D model for a relay.

1. Started in the Part Design workbench by creating a part.
2. Added two bodies, one for the main relay cube shape, and
3. one for modeling a single pin to be placement array'd.
4. I drew a chiral sketch mapped to the face on the underside of the relay.
5. I put a placement on the top face of the single pin, and
6. A placement on the first chiral shape I drew in the sketch for the relay cube.
7. I made an array of placements out of the chiral sketch, and then
8. used the Lattice Part Design pattern copy to attach the single pin to the array of placements.

That is what you see here:

[image]

[image]

I am trying to change the overall appearance of the relay to look like the example image shown here: https://www.digikey.com/products/en?keywords=EC2-3NU

But I cannot figure out how to change the appearance of the relay body and pins separately. Is there a way to do that?

DeepSOIC, 2020-05-03:
Hi!
Lattice2 Pattern feature does not respect appearance setting of the body it is in. You may have better success with your project if you avoid PartDesign, and use Part workflow instead. Lattice2 is actually made for Part workflow more than it is for PartDesign.

Also, Lattice2 was not made with shape colors in mind, so difficulties are to be expected unfortunately. (Lattice2 mostly uses colors to indicate if the object is an array of placements or an array of objects: orange is for placements, gray is for objects; gray is actually default shape color setting in freecad)

DeepSOIC, 2020-05-04:
after rereading your request, I realize that you are modeling the full shape of the relay as one PartDesign Body. PartDesign Body does not support feature colors, the body can only have one color. So, you should use Part workbench, at least for the last step of unifying the relay body with the pins.

yankee14, 2020-05-04:
Thanks for your help and suggestions. I've actually never used the Part workbench, so I'll give it a shot.

closed 2020-06-03 by yankee14