DeepSOIC/Lattice2 #49: Using LatticePattern with App::Link

Grawp, 2020-07-15:
Hello,
Is there a possibility to somehow at least indirectly via some workaround use linked PD body in LatticePattern? Trying to use it directly like normal PD body ends with `Feature Link is not a PartDesign feature. Unsupported.`
If it really can't be used right now would you consider adding a support for linked PD bodies in future?

DeepSOIC, 2020-07-15:
Why don't you use the original body? is it in a different file?
I certainly do want to invade Lattice2 with Link support (it will become Lattice3 in the process), but I don't have enough time and desire for it right now.

Grawp, 2020-07-15:
> Why don't you use the original body? is it in a different file?

Yep. I find if very practical to have a part/feature in a separate file. Mainly if it is being reused in many places.
You can also separately version such parts. It would be even more useful if those links could be parametrized.. possibly similarly to your ParaSeries (though I haven't actually tried ParaSeries just yet).

> I certainly do want to invade Lattice2 with Link support (it will become Lattice3 in the process), but I don't have enough time and desire for it right now.

Of course. I'll just squash all the parts to one file for now. Your WB is brilliant though. I couldn't imagine doing anything even a little more complex without it.

For example here's a mount for a flight controller supported by rubber dampers (the dampers are not shown):
[image]
Those 8 things with holes are where rubber dampers go and both their design and their placements (placed using your WB!) are driven by those yellow sketches. They are project specific.
But those 4 holes on the bottom with little locks inside? Those are holders for 2x3 dupont connector housings. They are done by **LatticePattern** and are candidates for reusing in **other projects** while also being candidates for being parametric on dupont housing size.

DeepSOIC, 2020-07-15:
Yes, i can easily understand why repeating a featureset body from another file is useful. Thanks for the detailed explanation, nonetheless. 

Zolko's Assembly4 is rather similar in its concept (but it is not array-centric like Lattice2), and has Link support at its core, you may want to check it out too.

sliptonic, 2020-09-22:
I'll second the desire for this.  I've been very interested in doing more design using these kinds of digital joinery (https://www.archdaily.com/797107/50-downloadable-digital-joints-for-woodworking)

It would be nice to be able to design the various joint features in separate files and then reuse them in the product design.