DeepSOIC/Lattice2 #52: ParaSeries fails when there's a link to a part in another document

LupinIII, 2021-12-23:
I am trying to generate a series of sketches with one varied parameter (diameter of a circle). I had no problem getting it to work in a "tryout" document. But when I did the same in a document where I actually wanted to use it, I get the following error message:

```19:14:48  Traceback (most recent call last):
  File "L:\*****\AppData\Roaming\FreeCAD\Mod\lattice2\lattice2BaseFeature.py", line 148, in execute
    plms = self.derivedExecute(obj)
  File "L:\*****\AppData\Roaming\FreeCAD\Mod\lattice2\lattice2ParaSeries.py", line 207, in derivedExecute
    doc2.copyObject(screen(selfobj.Object), True)
<class 'RuntimeError'>: {'sclassname': 'class Base::RuntimeError', 'sErrMsg': 'Document must be saved at least once before link to external objects', 'sfile': '', 'iline': 0, 'sfunction': '', 'swhat': 'Document must be saved at least once before link to external objects', 'btranslatable': False, 'breported': False}
```

Did take me quite some time and tries to find what causes this issue. But in the end it is the simple fact that a link to a body in another document breaks the ParaSeries. It does not matter where the link is in the document tree, neither does it matter where the ParaSeries is or if it has any relation to the linked object. Deleting all external links makes the ParaSeries work as expected.

OS: Windows 10 (10.0)
Word size of FreeCAD: 64-bit
Version: 0.20.Unknown
Build type: Release
Python version: 3.9.7
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.5.3
Locale: German/Austria (de_AT)

DeepSOIC, 2021-12-23:
thanks for the report! I'm not sure if i'll be able to fix it.

DeepSOIC, 2021-12-25:
Should be fixed, turned out to be really easy!
Please update Lattice2 and try again. If still not working, reopen the issue and maybe provide example files or step-by-step instruction.

closed 2021-12-25 by DeepSOIC

LupinIII, 2021-12-25:
Yes, thank you! That seems to have done it. It works in the document I had problems with earlier.

(nothing essential, almost OT from here on ;) )
That said, I have solved it in another way in this case already, but only because lattice2 is so versatile! I needed a cone of circles (for a CNC path) so I "revolved" a sketch of alternating standard and construction lines (all in a straight slanted line, alternating to avoid double edges in the revolution at every intermediate sketch point), then used lattice2's parametric downgrade to non-seam edges. In this case the result was the same as could have been achieved by a ParaSeries. (although as soon as I'd need to change the number of circles)

Too bad I only checked out lattice2 the first time two days ago. You can't imagine how often even the "simple" features like the parametric downgrade and subselect would have been useful to me, let alone the array features. I am a programmer (unfortunately not that much in Python), so everything that enables me to avoid manual repetition and automate things is right up my alley.

Thanks for the great work!

DeepSOIC, 2021-12-25:
great! 

ParaSeries and TopoSeries are very heavy tools, it's best to work around them if possible.

commit 67a297f1 (2021-12-25): ParaSeries, TopoSeries: fix #52 fail when there is a link