DeepSOIC/Lattice2 #59: With latest FC build Lattice shapestirng doesn't work

macdroid53, 2022-03-26:
`Running the Python command 'Lattice2_ShapeString' failed:
Traceback (most recent call last):
  File "/home/mac/.local/share/FreeCAD/Mod/lattice2/./lattice2ShapeString.py", line 285, in Activated
    CreateLatticeShapeString(name = "Strings")
  File "/home/mac/.local/share/FreeCAD/Mod/lattice2/./lattice2ShapeString.py", line 260, in CreateLatticeShapeString
    FreeCADGui.doCommand("f = lattice2ShapeString.makeLatticeShapeString(name='"+name+"')")
  File "<string>", line 1, in <module>
  File "/home/mac/.local/share/FreeCAD/Mod/lattice2/./lattice2ShapeString.py", line 77, in makeLatticeShapeString
    LatticeShapeString(obj)
  File "/home/mac/.local/share/FreeCAD/Mod/lattice2/./lattice2ShapeString.py", line 105, in __init__
    self.makeFoolObj(obj)
  File "/home/mac/.local/share/FreeCAD/Mod/lattice2/./lattice2ShapeString.py", line 142, in makeFoolObj
    self.draft_shape_string = _ShapeString(foolObj) 
  File "/home/mac/freecad-build/Mod/Draft/draftobjects/shapestring.py", line 59, in __init__
    obj.addProperty("App::PropertyBool", "MakeFace", "Draft", _tip).MakeFace = True

'NoneType' object has no attribute 'MakeFace'`

[code]
OS: Debian GNU/Linux 11 (bullseye) (XFCE/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 0.20.28431 (Git)
Build type: Unknown
Branch: master
Hash: 1d95f7b58da86406571377d4e0ce06190761d9b5
Python 3.9.2, Qt 5.15.2, Coin 4.0.0, OCC 7.5.1
Locale: C/Default (C)
Installed mods: 
  * kicadStepUpMod 10.11.6
  * Manipulator 1.4.3
  * DynamicData 2.45.0
  * fasteners 0.3.38
  * fcgear 1.0.0
  * plugins
  * Defeaturing
  * Assembly4 0.11.10
  * OpticsWorkbench 1.0.8
  * workfeature-macro
  * sheetmetal 0.2.49
  * A2plus 0.4.55a
  * CurvedShapes 1.0.1
  * Curves 0.3.0
  * lattice2 1.0.0
[/code]

DeepSOIC, 2022-03-26:
should be fixed by c6b5574e6ce702b91013bc42ad63a2ccba478409

closed 2022-03-26 by DeepSOIC

macdroid53, 2022-03-27:
Thanks again. Looks ok with today's source and a update from Addon manager.