DeepSOIC/Lattice2 #65: Lattice Icons have weird location in Linkstage.

HaasNL2, 2022-05-06:
I am working on an icon theme for Linkstage.
And it goes pretty good, but now there is a pattern Icon from Lattice2 on the part workbench. And since I like to have it complete I wanted to mod that one. But I couldn't.
It has a weird location link. I have no idea if it is an error in Linkstage or that it is the workbench. But maybe it rings you some bells if you see it. Normal icons don't have the :/icons/ in front of the location. And I guess the : breaks the iconset config file.

[image]

If you have Linkstage installed and want to see it yourself you can do as followed:
[image]

new icon:
[image]

DeepSOIC, 2022-05-07:
i don't understand icon locations, unfortunately. Here's an issue/PR on Part-o-magic: https://github.com/DeepSOIC/Part-o-magic/pull/54
Any help in sorting out this mess is appreciated.

DeepSOIC, 2022-05-07:
also, i think your new icon is conceptually incorrect (suggests that Lattice Pattern copies bodies, while it actually copies features).

HaasNL2, 2022-05-09:
> also, i think your new icon is conceptually incorrect (suggests that Lattice Pattern copies bodies, while it actually copies features).

I will add some changes, I tried the function first before making the icon but but you re right that it uses function, although it is hard to make it visible is such a small picture. I'll poke around for the icons, not a python programmer. But I compared path workbench and yours to see any difference but didn't.

HaasNL2, 2022-05-13:
Ok I made a mistake there was a typo it does work.

closed 2022-05-13 by HaasNL2