DeepSOIC/Lattice2 #68: Placement has to be attached to edge called Edge1

macdroid53, 2022-06-15:
When using the Placement if the support property is set to, for example Edge2, then Edge1 is ignored.

See Placement002 in this file. If the Support is changed to Edge1 the 3rd latch is populated. As it is, set to Edge2, no copy is populated for Edge1.

[Lattice2_Placement_on_Edge2.FCStd.zip](https://github.com/DeepSOIC/Lattice2/files/8909479/Lattice2_Placement_on_Edge2.FCStd.zip)

DeepSOIC, 2022-06-15:
Hi, this is by design, to allow cross-item attachment. For example, if you use Edge1 and Edge2 (which Lattice2 reinterprets as Child2.Edge1) as support, Lattice2 will loop as: Child1.Edge1+Clild2.Edge1, Child2.Edge1+Child3.Edge1. You can switch PlacementSubsequence.CycleMode to Periodic, then the first one will also be included (but appear as the last placement of the array).

DeepSOIC, 2022-06-15:
The shape of the sketch is a compound of wires, ChildX is a child of that compound. Each child in your sketch has just 1 edge. If you were to make the L-shape placer sketch, each child will have two edges.

macdroid53, 2022-06-15:
Excuse me, I've not looked at the way the sketch handles this is quite a while.

So, what associates the 2 edges of the L-shape to a child?

DeepSOIC, 2022-06-15:
Sketcher assembles edges into wires by looking at geometrically coincident vertices (the way you constrain it doesn't matter, actual positions do). Also, don't do triple (and higher) connections, they cause broken shapes and unstable edge/vertex numbering. (construction geometry is completely ignored, thankfully)

DeepSOIC, 2022-06-15:
Part::GeomLineSegment is the content of Geometry property. Lattice2 does not use it, it uses Shape property.

DeepSOIC, 2022-06-15:
Use Lattice2's Shape Info to examine shape structure, it was made for that.

```
Selected 1 subelements:
Vertex2: Vertex ([-14.905163, 35.62737, 4.0])

Selected document object:
  Name = Sketch002
  Label = Sketch002
  Is placement/array = False
Structure: 
Compound (3 objects):
    Wire (1 segments)
    Wire (1 segments)
    Wire (1 segments)
```

macdroid53, 2022-06-15:
Aha, I see in my debugger the SubShape.SubShapes has the association between two of the edges.. But, what process made that association?

I'll have a look at the Shape Info here shortly.

macdroid53, 2022-06-15:
Ok, so that's a sort of base process.

Thanks...I hope I haven't asked too many questions. I appreciate your patience.

DeepSOIC, 2022-06-15:
well, they are good questions!

quiret, 2022-10-27:
Thank you, DeepSOIC, for creating this amazing workbench! You are a technical mastermind of which there are only few. Your workbench enables workflows that are complex but which are too cumbersome to do in regular FreeCAD. I also love that you added the PartDesign equivalent of Feature Mirror as Lattice2 Mirror tool! You enable people.