DeepSOIC/Lattice2 #77: Additive helix not working

SarenT, 2023-12-21:
I have a feature with an additive helix, which I would like to pattern on another body. It turns out to be without the helix. The feature looks fine alone, the patterned feature looks fine except for the helix.
[PatternDesignLattice2_Issue.FCStd.zip](https://github.com/DeepSOIC/Lattice2/files/13734221/PatternDesignLattice2_Issue.FCStd.zip)
[image]

DeepSOIC, 2023-12-21:
after a quick look into your example, i'm not quite sure what's going on there. Switching "SelfIntersect" property of LatticePattern001 to "true" fixes it for me, but it should in theory work with selfintersect turned off...

SarenT, 2023-12-22:
Thanks a lot, indeed it fixed for me too.
[image]