DeepSOIC/Lattice2 #78: Polar Array count is uneditable in latest freecad weekly build

alpdigitech, 2024-01-15:


DeepSOIC, 2024-01-30:
works fine for me in: 
```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35735 (Git)
Build type: Release
Branch: main
Hash: c12a5c2a163c1a277cf273106899209792be152d
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: English/Germany (en_DE)
Installed mods: 
  * CurvesWB 0.4.4
  * DeepConsole
  * Defeaturing 1.2.0
  * fir_design
  * freecad.gears 1.0.0
  * Lattice2 1.0.0
  * Part-o-magic 1.1.0
  * TabBar
  * _disabled
```
need more details.

Can it be that you've created the array in Span/Step mode? then, Count should be uneditable by design.

DeepSOIC, 2025-11-01:
closing because can't reproduce.