DeepSOIC/Lattice2 #80: Object of type ViewProviderDocumentObject is not JSON serializable

numo68, 2024-02-15:
Not sure whether this is FreeCAD or Lattice2 problem, so I try here first:

Lattice2: v1.0.0 installed through Addon Manager

FreeCAD on a MacBook Air M2:

```
OS: macOS 14.2.1
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * freecad.gears 1.0.0
  * fasteners 0.5.3
  * lattice2 1.0.0
```

Project: [Wago221-2411-Clip.FCStd.zip](https://github.com/DeepSOIC/Lattice2/files/14301682/Wago221-2411-Clip.FCStd.zip)

Saving the project generates:

```
20:54:09  PropertyPythonObject::toString(): failed for <class 'lattice2AttachablePlacement.ViewProviderAttachablePlacement'>
20:54:09  Traceback (most recent call last):
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.10/json/__init__.py", line 231, in dumps
    return _default_encoder.encode(obj)
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.10/json/encoder.py", line 199, in encode
    chunks = self.iterencode(o, _one_shot=True)
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.10/json/encoder.py", line 257, in iterencode
    return _iterencode(o, 0)
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.10/json/encoder.py", line 179, in default
    raise TypeError(f'Object of type {o.__class__.__name__} '
<class 'TypeError'>: Object of type ViewProviderDocumentObject is not JSON serializable
20:54:09  PropertyPythonObject::toString(): failed for <class 'lattice2AttachablePlacement.ViewProviderAttachablePlacement'>
20:54:09  Traceback (most recent call last):
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.10/json/__init__.py", line 231, in dumps
    return _default_encoder.encode(obj)
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.10/json/encoder.py", line 199, in encode
    chunks = self.iterencode(o, _one_shot=True)
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.10/json/encoder.py", line 257, in iterencode
    return _iterencode(o, 0)
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.10/json/encoder.py", line 179, in default
    raise TypeError(f'Object of type {o.__class__.__name__} '
<class 'TypeError'>: Object of type ViewProviderDocumentObject is not JSON serializable
20:54:09  PropertyPythonObject::toString(): failed for <class 'lattice2AttachablePlacement.ViewProviderAttachedPlacementSubsequence'>
20:54:09  Traceback (most recent call last):
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.10/json/__init__.py", line 231, in dumps
    return _default_encoder.encode(obj)
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.10/json/encoder.py", line 199, in encode
    chunks = self.iterencode(o, _one_shot=True)
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.10/json/encoder.py", line 257, in iterencode
    return _iterencode(o, 0)
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.10/json/encoder.py", line 179, in default
    raise TypeError(f'Object of type {o.__class__.__name__} '
<class 'TypeError'>: Object of type ViewProviderDocumentObject is not JSON serializable
20:54:09  PropertyPythonObject::toString(): failed for <class 'lattice2PDPattern.ViewProviderLatticePDPattern'>
20:54:09  Traceback (most recent call last):
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.10/json/__init__.py", line 231, in dumps
    return _default_encoder.encode(obj)
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.10/json/encoder.py", line 199, in encode
    chunks = self.iterencode(o, _one_shot=True)
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.10/json/encoder.py", line 257, in iterencode
    return _iterencode(o, 0)
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.10/json/encoder.py", line 179, in default
    raise TypeError(f'Object of type {o.__class__.__name__} '
<class 'TypeError'>: Object of type ViewProvider is not JSON serializable
```

The saved file still loads okay but I did not examine it thoroughly for subtle damage.

The message seems to be triggered by upgrading the FreeCAD from 0.21.1 to 0.21.2.

Thanks

DeepSOIC, 2024-02-15:
well, apparently, FreeCAD is now using json instead of pickle. For pickle, i have 
```
    def __getstate__(self):
        return None

    def __setstate__(self,state):
        return None
```
that is supposed to prevent these errors. For json however, it seems like i can't override the serializer... I guess i have to dig fc forum a bit to figure out how to fix it, lattice2 can't be the only addon that is affected...

numo68, 2024-02-22:
> well, apparently, FreeCAD is now using json instead of pickle.

Digging through docs and texts they seem to be doing that for quite a long time, I have found mentioning it in some text for 0.19-dev. So maybe they just disabled something now.

I don't know anything about the internal structures work or how addons are developed. However, https://wiki.freecad.org/Scripted_objects suggests overriding `dumps()` and `loads()` . See also the end of the https://forum.freecad.org/viewtopic.php?t=40812 discussion.

Connor9220, 2024-04-17:
Is there a fix for this yet?  I'm seeing this with latest Lattice2 and 0.21.2

DeepSOIC, 2024-04-19:
should be fixed now, enjoy =)
Sorry, i'm bad at maintaining my workbenches well these days...

closed 2024-04-19 by DeepSOIC

Connor9220, 2024-04-20:
> should be fixed now, enjoy =) Sorry, i'm bad at maintaining my workbenches well these days...

Thanks!  And I understand.  Hard to keep up with stuff.

commit 16eb1073 (2024-04-19): fix #80 "not JSON serializable" errors in FC0.21.2