DeepSOIC/Lattice2 #81: Possible placement error

pyro9, 2024-02-20:
I ran into what looks like a minor placement error with a zero radius polar array. I created a polar array at the origin with 0 radius, then populated it with a fanfold shaped loft (a surface) with it's vertex at the origin. Then I use the resulting object to split apart a cylinder. I would expect 2 slices. but I get many slices. Further, both "halves" of the cylinder are in the same slice.

If I instead apply a draft polar array to the loft, I get the expected results when I use it to slice apart the cylinder.

The error seems to be near the origin since if I instead slice apart a ring, I get the expected results with either method.

I made a short [video](https://www.youtube.com/watch?v=snhc86sTqv4) showing the issue.

DeepSOIC, 2024-02-20:
I don't have your project, but from what i see in the video i think i can identify the problem. For a shape to be usable for a boolean operation, there should not be any subshapes that intersect or coincide - they must be then shared. Lattice2 arrays doesn't automatically set up this sharing (it just makes a Part Compound of the moved copies). 

Try applying Part Union to "Populate Polar array with Loft" object, and use the union as a slice tool. (Part Connect and Part BooleanFragments may also do the trick of connecting up the shape).

pyro9, 2024-02-24:
I can confirm that Union does fix the issue. What confused me is that when I slice a ring rather than a cylinder with the populated array, it works as expected yielding only two slices, as shown [here](https://youtu.be/GEEjPzwmWcw?t=793).