DeepSOIC/Lattice2 #86: Error in current 0.22 development versions: Property container has no property 'Support'

johnsonm, 2024-05-02:
I have a project using Lattice2, and in current weekly builds I get:

```
21:38:39  pyException: Traceback (most recent call last):
  File "/var/home/johnsonm/.local/share/FreeCAD/Mod/lattice2/./lattice2BaseFeature.py", line 278, in onDocumentRestored
    self.disableAttacher(selfobj)
  File "/var/home/johnsonm/.local/share/FreeCAD/Mod/lattice2/./lattice2BaseFeature.py", line 272, in disableAttacher
    selfobj.setEditorMode(prop, 0 if enable else 2)
<class 'AttributeError'>: Property container has no property 'Support'
```

I previously reported this in https://github.com/FreeCAD/FreeCAD/issues/13746 and was pointed to https://github.com/FreeCAD/FreeCAD/pull/12714 where it was made clear that this breaking change was intentional to fix a real bug.

I suspect this is easy to reproduce, but merely loading CubeSet.FCStd from  https://gitlab.com/mcdanlj/RSVPGameReplacementParts in the current weekly build is sufficient to demonstrate the problem.

(Because I'm a relatively new user of Lattice2 I imagine that I've made some mistakes in how I used it.)

Thanks for Lattice2! ☺

johnsonm, 2024-05-03:
As far as I can tell, this means changing `Support` to `AttachmentSupport` to support 0.22 dev, but that wouldn't be backward compatible with older FreeCAD versions. (At least, making that change avoids the errors with current FreeCAD dev build.) But I don't know enough about the intent of that to know whether it's a correct change.

DeepSOIC, 2024-05-03:
yes, that's quite a pain.

I'm struggling to find time to maintain lattice2, so i recommend using stable freecad for now... i'll get to it eventually.

johnsonm, 2024-05-04:
I have now discovered that stable freecad Appimage doesn't display properly on my system, but with the PR I just pushed, I at least don't see any relevant errors on the console, and the pattern is simple enough that I believe my fix is right.