DeepSOIC/Lattice2 #89: Issue in FreeCAD 0.22 dev version

onekk, 2024-06-01:
Hello, there is an issue on the new 0.22 dev version:
here the relevant informations:

```
Running the Python command 'Lattice2_ShapeString' failed:
Traceback (most recent call last):
  File "/Users/ed/Library/Application Support/FreeCAD/Mod/lattice2/./lattice2ShapeString.py", line 298, in Activated
    CreateLatticeShapeString(name = "Strings")
  File "/Users/ed/Library/Application Support/FreeCAD/Mod/lattice2/./lattice2ShapeString.py", line 273, in CreateLatticeShapeString
    FreeCADGui.doCommand("f = lattice2ShapeString.makeLatticeShapeString(name='"+name+"')")
  File "<string>", line 1, in <module>
  File "/Users/ed/Library/Application Support/FreeCAD/Mod/lattice2/./lattice2ShapeString.py", line 77, in makeLatticeShapeString
    LatticeShapeString(obj)
  File "/Users/ed/Library/Application Support/FreeCAD/Mod/lattice2/./lattice2ShapeString.py", line 106, in __init__
    self.makeFoolObj(obj)
  File "/Users/ed/Library/Application Support/FreeCAD/Mod/lattice2/./lattice2ShapeString.py", line 143, in makeFoolObj
    self.draft_shape_string = _ShapeString(foolObj) 
                              ^^^^^^^^^^^^^^^^^^^^^
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/Draft/draftobjects/shapestring.py", line 48, in __init__
    self.set_properties(obj)
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/Draft/draftobjects/shapestring.py", line 52, in set_properties
    properties = obj.PropertiesList
                 ^^^^^^^^^^^^^^^^^^

'FoolFeatureDocumentObject' object has no attribute 'PropertiesList'
```

And the FreeCAD forum discussion thread:

https://forum.freecad.org/viewtopic.php?t=88089

Kind Regards

Carlo D.

Roy-043, 2024-06-01:
See this forum post for a possible solution:
https://forum.freecad.org/viewtopic.php?t=88089#p763098

spkane, 2024-07-18:
@DeepSOIC Have you had an opportunity to look at this yet?

I am curious if there are any thoughts about the approach forward and a guesstimated timeline. Thanks!

DeepSOIC, 2024-07-18:
I have looked at it once, and couldn't knock out a good solution by the time the time slot ended... It is not that involved, but i am having a hard time convincing myself to work on it in spite of all the other stuff i want/have to do.

jrb0411, 2024-11-11:
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Curves 0.6.48
  * ExplodedAssembly
  * lattice2 1.0.0
  * Woodworking 0.21.2.33771

ShapeString Error
08:42:37  Running the Python command 'Lattice2_ShapeString' failed:
Traceback (most recent call last):
  File "C:\Users\jbrowning\AppData\Roaming\FreeCAD\Mod\lattice2\.\lattice2ShapeString.py", line 298, in Activated
    CreateLatticeShapeString(name = "Strings")
  File "C:\Users\jbrowning\AppData\Roaming\FreeCAD\Mod\lattice2\.\lattice2ShapeString.py", line 273, in CreateLatticeShapeString
    FreeCADGui.doCommand("f = lattice2ShapeString.makeLatticeShapeString(name='"+name+"')")
  File "<string>", line 1, in <module>
  File "C:\Users\jbrowning\AppData\Roaming\FreeCAD\Mod\lattice2\.\lattice2ShapeString.py", line 77, in makeLatticeShapeString
    LatticeShapeString(obj)
  File "C:\Users\jbrowning\AppData\Roaming\FreeCAD\Mod\lattice2\.\lattice2ShapeString.py", line 106, in __init__
    self.makeFoolObj(obj)
  File "C:\Users\jbrowning\AppData\Roaming\FreeCAD\Mod\lattice2\.\lattice2ShapeString.py", line 143, in makeFoolObj
    self.draft_shape_string = _ShapeString(foolObj) 
                              ^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\FreeCAD 1.0\Mod\Draft\draftobjects\shapestring.py", line 48, in __init__
    self.set_properties(obj)
  File "C:\Program Files\FreeCAD 1.0\Mod\Draft\draftobjects\shapestring.py", line 52, in set_properties
    properties = obj.PropertiesList
                 ^^^^^^^^^^^^^^^^^^

'FoolFeatureDocumentObject' object has no attribute 'PropertiesList'

pyro9, 2025-01-25:
#96 fixes this.

closed 2025-04-14 by DeepSOIC