DeepSOIC/Lattice2 #91: Polar array with para series doesn't copy over placement of compound

SpeeQz1, 2024-11-06:
When trying to create a polar array from para series the placement of the objects doesn't get copied unlike with Linear array where they do get copied.

[image]

[image]

Tutorial used for reference:
[Cogset Tutorial](https://github.com/DeepSOIC/Lattice2/wiki/Cogset-Tutorial)

FreeCAD-Bundle AppImage: weekly-build
OS: Linux Mint 22 ( Cinnamon )

DeepSOIC, 2024-11-06:
as i said, if you're using "build array", the offset is canceled out. That is by design. That's basically the whole point of "build array", the object as-is is assumed to be pre-placed. Just select ParaSeries and Polar Array and apply "Populate with children" and you get what you want. Or select ParaSeries, + Custom + Polar Array and apply "move children", that will also work.

DeepSOIC, 2024-11-06:
Also, your "Populate Custom with Body" is redundant, you should be able to apply paraseries directly to a body.

SpeeQz1, 2024-11-06:
"Populate with Children" and para series from just body worked for me and seems to have properly copied over the elements thanks.
I don't know how I missed the polar array working with the para series when using the first populate with children option, so sorry for the trouble.

Here are the results of my testing:

1. Custom + Para Series + Linear Array:
* Populate with Children = Objects are properly copied to each kite
* Populate with Children: Build Array = Objects are properly copied to each kite
2. Custom + Para Series + Polar Array:
* Populate with Children = Objects are properly copied to each kite
* Populate with Children: Build Array = Objects are __NOT__ copied to each kite and stay in the middle
3. Body + Para Series + Linear Array:
* Populate with Children = Objects are properly copied to each kite
* Populate with Children: Build Array = Objects are properly copied to each kite
4. Body + Para Series + Polar Array:
* Populate with Children = Objects are properly copied to each kite
* Populate with Children: Build Array = Objects are __NOT__ copied to each kite and stay in the middle

closed 2024-11-21 by SpeeQz1