DeepSOIC/Lattice2 #92: FR: Array spacing based on bounding boxes

SpeeQz1, 2024-11-21:
I hope that I haven't missed such a feature already but it would be nice to be able to change the spacing between elements of an array populated with the children of a para series based on the bounding box and not the center of the object.

(In the picture are the different Linear Array options)
[image]

DeepSOIC, 2024-11-22:
maybe i'll do something some day. As of now,if you know the sizes of bounding boxes upfront, you can set up a linear array with custom values, either entered directly into property editor, or to be fetched from a spreadsheet.

SpeeQz1, 2024-11-22:
@DeepSOIC doesn't the linear array use only a single value for the step and not an array of values?

DeepSOIC, 2024-11-22:
it can do both. Set `Values Source` property to "Values Property", and then use `Values` property to enter the individual positions along the line. Similarly, you can set it up to pull a column from a spreadsheet.