DeepSOIC/Lattice2 #94: Part Connect failing to merge wires together into a singular one

SpeeQz1, 2024-11-26:
Following the tutorial [Cogset Tutorial](https://github.com/DeepSOIC/Lattice2/wiki/Cogset-Tutorial) and making a very basic example, I couldn't get the wires to connect as they kept failing due to errors:

[image]
[image]
[image]

```sh
11:38:58  pyException: Traceback (most recent call last):
  File "/tmp/.mount_freecaiKaaAK/usr/Mod/Part/BOPTools/JoinFeatures.py", line 129, in execute
    rst = JoinAPI.connect([obj.Shape for obj in selfobj.Objects], selfobj.Tolerance)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_freecaiKaaAK/usr/Mod/Part/BOPTools/JoinAPI.py", line 71, in connect
    dim = ShapeMerge.dimensionOfShapes(list_of_shapes)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_freecaiKaaAK/usr/Mod/Part/BOPTools/ShapeMerge.py", line 231, in dimensionOfShapes
    raise TypeError("Shapes are of different dimensions ({t1} and {t2}), and cannot be merged or compared.".format(t1= list_of_shapes[0].ShapeType, t2= sht))
<class 'TypeError'>: Shapes are of different dimensions (Wire and Vertex), and cannot be merged or compared.
11:38:58  Connect: Shapes are of different dimensions (Wire and Vertex), and cannot be merged or compared.
```

---

```
OS: Linux Mint 22 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods: 
  * fasteners 0.5.31 (Disabled)
  * lattice2 1.0.0
  * Manipulator 1.5.7
  * parts_library
  * Alternate_OpenSCAD 1.0.0
  * CurvedShapes 1.0.13
  * FreeCAD-Ribbon 1.4.8.1
  * SearchBar 1.0.2
  * OpenTheme 2024.9.1
```

DeepSOIC, 2024-11-26:
according to the error message, the sketch has some stray vertices. From your screenshots, it looks like there is one at sketch origin that is not construction geometry. Try deleting it, or toggle into being construction geometry.

SpeeQz1, 2024-11-26:
> according to the error message, the sketch has some stray vertices. From your screenshots, it looks like there is one at sketch origin that is not construction geometry. Try deleting it, or toggle into being construction geometry.

had to remake the sketch but I checked the vertices and there isn't any free one yet the part fails again:

[image]
[image]
[image]

```
19:17:42  pyException: Traceback (most recent call last):
  File "/tmp/.mount_freecaLIcAEd/usr/Mod/Part/BOPTools/JoinFeatures.py", line 129, in execute
    rst = JoinAPI.connect([obj.Shape for obj in selfobj.Objects], selfobj.Tolerance)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_freecaLIcAEd/usr/Mod/Part/BOPTools/JoinAPI.py", line 71, in connect
    dim = ShapeMerge.dimensionOfShapes(list_of_shapes)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_freecaLIcAEd/usr/Mod/Part/BOPTools/ShapeMerge.py", line 231, in dimensionOfShapes
    raise TypeError("Shapes are of different dimensions ({t1} and {t2}), and cannot be merged or compared.".format(t1= list_of_shapes[0].ShapeType, t2= sht))
<class 'TypeError'>: Shapes are of different dimensions (Wire and Vertex), and cannot be merged or compared.
19:17:42  Connect: Shapes are of different dimensions (Wire and Vertex), and cannot be merged or compared.
```

DeepSOIC, 2024-11-27:
I guess, it might be some new behavior of FreeCAD here that i'm not quite familiar with. I can open your file in my older FreeCAD, and change all the line ends into construction,  and then the populate can be connected.
[image]

If that still doesn't work for you, Lattice2 *Downgrade to Wires* then *Connect* works.

SpeeQz1, 2024-11-27:
Both solutions seemed to have worked so thanks. But it would be good to take a look into somehow fixing this so that normal points can be used.

DeepSOIC, 2025-04-14:
So, in FC v1.0, standalone vertices created in Sketcher follow the the state of construction mode toggle, just like all other geometry.
I'm closing this as it is not a Lattice2 issue; i might even say it is not a bug at all. Basic features like Part Extrude are also affected by these vertices.

closed 2025-04-14 by DeepSOIC