DeepSOIC/Lattice2 #98: Lattice Shape String breaks when changing strings (Freecad 1.0)

Zetabite, 2025-04-12:
When changing the default strings to anything else, the Lattice Shape String breaks with this error/warning:
```
16:48:55  ShapeString: string has no wires
16:48:55  pyException: Traceback (most recent call last):
  File "/home/zetabite/.local/share/FreeCAD/Mod/lattice2/./lattice2ShapeString.py", line 222, in execute
    shape.Placement = App.Placement(alignPnt*(-1.0), App.Rotation()).multiply(shape.Placement)
<class 'AttributeError'>: 'NoneType' object has no attribute 'Placement'
16:48:55  Strings: 'NoneType' object has no attribute 'Placement'
```

DeepSOIC, 2025-04-14:
can't confirm, works here.
```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/Germany (en_DE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Light/Fusion
Installed mods: 
  * CurvesWB 0.4.4
  * DeepConsole
  * fir_design
  * Lattice2 1.0.0
  * Part-o-magic 1.1.0
```
can you attach a project with a broken shapestring?

Zetabite, 2025-04-14:
[Lattice2BrokenString.FCStd.zip](https://github.com/user-attachments/files/19742658/Lattice2BrokenString.FCStd.zip)

Zetabite, 2025-04-14:
OS: Arch Linux (xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git)
Build type: Release
Branch: makepkg
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.13.2, Qt 6.9.0, Coin 4.0.3, Vtk 9.4.2, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * lattice2 1.0.0

DeepSOIC, 2025-04-14:
i see it happening, will investigate.

DeepSOIC, 2025-04-14:
should be fixed

[image]

closed 2025-04-14 by DeepSOIC

Zetabite, 2025-04-14:
It's now broken in another way. It now tries to cramp them all on one side:

[image]

DeepSOIC, 2025-04-14:
i've seen this. This is because Extrude tries to interpret all the mess as one set of wires that form some faces with some holes. 

One solution is to set `ShapeStrings.MakeFace = true`, then Extrude will not use its face filling algorithm and just do it. 

Another is to populate ShapeStrings first, then extrude - it should work if the letters don't overlap. 

And another is to create the shapestrings to use an array from the start (select the PolarArray object, then click ShapeStrings button). This is almost just a shortcut for the previous one.

Zetabite, 2025-04-14:
Worked like a charm. Thank you very much. Your tool really helped me getting my project to a finish line :D

commit f0c0f64e (2025-04-14): ShapeStrings: fix #98 error with whitespace-only strings